sharandac/My-TTGO-Watch #329: Buttons are not reacting.

MarinGHG, 2022-01-06:
I installed the newest Version of the My-TTGO Watch and loaded it up but when I open an app, I can't click return anymore. If I press return to close the app i get a haptic feedback but nothing happens other buttons on the keyboard for example work fine. I'm using a TTGO T-watch v2.

sharandac, 2022-01-07:
Thanks for your feedback! The return/enter button is intended for multi-line entries and has no meaning for single-line entries. The check mark at the bottom right is intended to end/accept the input.

MarinGHG, 2022-01-07:
I have both, the TTGO T-watch v1 and v2. But I only get this Error on the v2, both are running on the same version.

sharandac, 2022-01-08:
Ok, thanks for the information. Can you tell me exactly what you did and in which app it occurs so I can track the error exactly?

Crsarmv7l, 2022-01-09:
I can provide only one specific instance of a button not working.

In utilities, power off only vibrates and does not power off.

I didn't deem it important enough to file an issue since that was my only instance.

MarinGHG, 2022-01-09:
For me, all buttons that are there to close an app are not working.

MarinGHG, 2022-01-09:
[image]

MarinGHG, 2022-01-09:
I can Exit Apps by Just Tuning the watch Off and on

closed 2022-01-27 by sharandac