sharandac/My-TTGO-Watch #331: Why is there 3 problems always coming when i try to run it in my TTGO

xXDevboiXx, 2022-01-10:
3 errors always appear and i cant seem to ever put it in my watch, i use PlatformIO (VSC Of course) i have the 2020 model, help please

sharandac, 2022-01-10:
Thank you for your feedback. But without a reasonably precise description of what an "error" is and where it occurs, we have no way of helping you.

Crsarmv7l, 2022-01-11:
> 3 errors always appear and i cant seem to ever put it in my watch, i use PlatformIO (VSC Of course) i have the 2020 model, help please

Watch this and please follow step by step
https://www.youtube.com/watch?v=wUGADCnerCs

Chances are it is a config issue, or you didn't select the correct target to build for (the firmware supports v1, v2, v3 2021, M5paper and a few others I believe)

sharandac, 2022-01-11:
#321

xXDevboiXx, 2022-01-12:
> Thank you for your feedback. But without a reasonably precise description of what an "error" is and where it occurs, we have no way of helping you.

Would you like a screenshot?

xXDevboiXx, 2022-01-12:
> Watch this and please follow step by step https://www.youtube.com/watch?v=wUGADCnerCs
>
> Chances are it is a config issue, or you didn't select the correct target to build for (the firmware supports [...]

I chose (and want) V3 and abt the video, i watched it and followed it in order but still dosent work... 1 of the errors was smthn abt the GPS...

sharandac, 2022-01-12:
V3 doesn't have a GPS at all?

Crsarmv7l, 2022-01-12:
> I chose (and want) V3 and abt the video, i watched it and followed it in order but still dosent work... 1 of the errors was smthn abt the GPS...

You could post the build log, because like was said earlier guessing gets nowhere.

xXDevboiXx, 2022-01-15:
> V3 doesn't have a GPS at all?

Thats weird bcoz it said smthn abt the gps void or smthn. like i said before, would you like a screenshot

xXDevboiXx, 2022-01-22:
> You could post the build log, because like was said earlier guessing gets nowhere.
Update, used v2 compiling worked, but heres the thing, the uploadings not working build log:
[image]

xXDevboiXx, 2022-01-22:
update 2, 1 error, i think the problem is the upload_port function, and the silcon labs driver isnt working?

xXDevboiXx, 2022-01-22:
> update 2, 1 error, i think the problem is the upload_port function, and the silcon labs driver isnt working?
(the error in case you need it:Error: Please specify `upload_port` for environment or use global `--upload-port` option. )

Crsarmv7l, 2022-01-22:
> > > > > 3 errors always appear and i cant seem to ever put it in my watch, i use PlatformIO (VSC Of course) i have the 2020 model, help please
> > > > 
> > > > 
> > > > Watch this and please follow step by step https://www.youtube.com/watch?v=wUGADCnerCs
> > > > Chances are it is a config issue, or you didn't select the correct target to build for (the firmware supports v1, v2, v3, 2021, M5paper and a few others I believe)
> > > 
> > > 
> > > I chose (and want) V3 and abt the video, i watched it and followed it in order but still dosent work... 1 of the errors was smthn abt the GPS...
> > 
> > 
> > You could post the build log, because like was said earlier guessing gets nowhere.
> > Update, used v2 compiling worked, but heres the thing, the uploadings not working build log:
> > [image]
> > had to censor a bunch of stuff for protection, i can give you the build log in text (removing private information) if you need

Those are known warnings. I have them as well they don't impact a successful build.

Crsarmv7l, 2022-01-22:
> > update 2, 1 error, i think the problem is the upload_port function, and the silcon labs driver isnt working?
> > (the error in case you need it:Error: Please specify `upload_port` for environment or use global `--upload-port` option. )

Dude....I know that is covered in the video tutorial.

xXDevboiXx, 2022-01-22:
> Those are known warnings. I have them as well they don't impact a successful build.

so why is it impacting my upload?

xXDevboiXx, 2022-01-22:
> > > > > > > 3 errors always appear and i cant seem to ever put it in my watch, i use PlatformIO (VSC Of course) i have the 2020 model, help please
> > > > > > 
> > > > > > 
> > > > > > Watch this and please follow step by step https://www.youtube.com/watch?v=wUGADCnerCs
> > > > > > Chances are it is a config issue, or you didn't select the correct target to build for (the firmware supports v1, v2, v3, 2021, M5paper and a few others I believe)
> > > > > 
> > > > > 
> > > > > I chose (and want) V3 and abt the video, i watched it and followed it in order but still dosent work... 1 of the errors was smthn abt the GPS...
> > > > 
> > > > 
> > > > You could post the build log, because like was said earlier guessing gets nowhere.
> > > > Update, used v2 compiling worked, but heres the thing, the uploadings not working build log:
> > > > [image]
> > > > had to censor a bunch of stuff for protection, i can give you the build log in text (removing private information) if you need
> > 
> > 
> > Those are known warnings. I have them as well they don't impact a successful build.
> 
> question, how do i identify if its the ttgo watch or not will it have esp 32 written or smthn

xXDevboiXx, 2022-01-22:
> > > update 2, 1 error, i think the problem is the upload_port function, and the silcon labs driver isnt working?
> > > (the error in case you need it:Error: Please specify `upload_port` for environment or use global `--upload-port` option. )
> 
> Dude....I know that is covered in the video tutorial.
> 
> Edit: Ok let me be more helpful. Open a terminal in PlatformIO. the command "pio device list" should show your device. If it doesn't you messed up the driver install. You can also check it in Quick Access Menu > PIO Home > Devices, which will inform you that it should be automatically detected, the COM port and other info, but you can manually set it in the project Platformio.ini if you know the COM port. You can also set it via the cli see here: https://docs.platformio.org/en/latest/core/userguide/cmd_run.html
> 
> You messed up installing the usb driver most likely.

but heres the thing, it wasnt there.

Crsarmv7l, 2022-01-22:
> so why is it impacting my upload?

They aren't. 

Somewhere in the build files it establishes those as "should be included" but they aren't there. This doesn't actually affect anything as nothing is needed from those links. 

Example: Say you generate a shortcut icon to a word document. Then you move that word document to your desktop. You can still open the word doc from your desktop right, but your shortcut link wont work anymore. Does not being able to use the shortcut mean the word doc is gone or just moved? Same thing.

Most of those were probably for old code that has been removed. Nimble for instance is related to the BLE stack, but having looked through the BLE extensively I haven't come across anything that seems to use it.

It is your usb driver probably

xXDevboiXx, 2022-01-23:
> They aren't.
>
> Somewhere in the build files it establishes those as "should be included" but they aren't there. This doesn't actually affect anything as nothing is needed from those links.
>
> Example: [...]

ok, everything worked, it was fine, soooo i tried to install v1 and im missing a file which is not included a file that has to belinked

xXDevboiXx, 2022-01-23:
ALRIGHT I AM PLEASE TO SAY THAT THIS ISSUE IS finished-

closed 2022-01-23 by xXDevboiXx