sharandac/My-TTGO-Watch #332: It dosen't work on my t-watch-2021

ILXR, 2022-01-11:
I checkout to branch 'experimental' and build and upload successfully.
But my watch is always black ( it flashes every time i restart )...
And there's the output in Serial Monitor:

`
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 271414342, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:936
load:0x40078000,len:10124
load:0x40080400,len:5828
entry 0x400806a8
[I][hardware.cpp:169] hardware_setup(): I2C device at: 0x0d
[I][hardware.cpp:169] hardware_setup(): I2C device at: 0x15
[I][hardware.cpp:169] hardware_setup(): I2C device at: 0x18
[I][hardware.cpp:169] hardware_setup(): I2C device at: 0x51
[I][callback.cpp:92] callback_init(): init callback_t structure success for: powermgm
[I][callback.cpp:92] callback_init(): init callback_t structure success for: button
[I][callback.cpp:92] callback_init(): init callback_t structure success for: powermgm loop
[I][basejsonconfig.cpp:135] load(): reading json failed, call defaults, file: /motor.json
[I][basejsonconfig.cpp:135] load(): reading json failed, call defaults, file: /display.json
[I][framebuffer.cpp:160] framebuffer_setup(): framebuffer: 0x0x3ffd4340 (4800 bytes, 240x240px)
[E][callback.cpp:199] callback_send(): no callback structure found
[E][callback.cpp:199] callback_send(): no callback structure found
[I][splashscreen.cpp:74] splash_screen_stage_one(): use default boot logo
[W][SPIFFS.cpp:71] begin(): SPIFFS Already Mounted!
[I][basejsonconfig.cpp:135] load(): reading json failed, call defaults, file: /pmu.json
[I][callback.cpp:92] callback_init(): init callback_t structure success for: blectl
[I][basejsonconfig.cpp:135] load(): reading json failed, call defaults, file: /bma.json
[I][basejsonconfig.cpp:135] load(): reading json failed, call defaults, file: /wificfg.json
[I][esp32-hal-i2c.c:1138] i2cProcQueue(): Bus busy, reinit
[I][callback.cpp:92] callback_init(): init callback_t structure success for: timesync
[I][basejsonconfig.cpp:123] load(): json config deserializeJson() success: Ok, file: /rtcctr.json
[I][basejsonconfig.cpp:354] save(): json config serializeJson() success: /rtcctr.json
[E][callback.cpp:199] callback_send(): no callback structure found
[I][basejsonconfig.cpp:135] load(): reading json failed, call defaults, file: /timesync.json
[I][callback.cpp:92] callback_init(): init callback_t structure success for: wifictl
[I][pcf8563.cpp:333] syncToSystem(): syncToSystem: 2022 1 11 - 7 38 27

[I][pcf8563.cpp:343] syncToSystem(): syncToSystem: 122 1 11 - 7 38 27 

[I][timesync.cpp:280] timesyncToSystem(): TZ rule: GMT0
[I][callback.cpp:92] callback_init(): init callback_t structure success for: pmu
[I][basejsonconfig.cpp:135] load(): reading json failed, call defaults, file: /blectl.json
[I][hardware.cpp:235] hardware_setup(): Total heap: 266596
[I][hardware.cpp:236] hardware_setup(): Free heap: 205324
[I][hardware.cpp:237] hardware_setup(): Total PSRAM: 0
[I][hardware.cpp:238] hardware_setup(): Free PSRAM: 0
[I][gui.cpp:90] gui_setup(): mainbar setup
[I][callback.cpp:92] callback_init(): init callback_t structure success for: rtctl
[I][gui.cpp:95] gui_setup(): mainbar tile setup
[E][main_tile.cpp:379] main_tile_update_time(): maintile not initialized
[I][callback.cpp:92] callback_init(): init callback_t structure success for: sensor
[I][callback.cpp:92] callback_init(): init callback_t structure success for: styles
[I][gui.cpp:97] gui_setup(): app tile setup
[I][gui.cpp:99] gui_setup(): note tile setup
[I][gui.cpp:101] gui_setup(): setup tile setup
[I][setup_tile.cpp:101] setup_tile_setup(): setup tile finish
[I][gui.cpp:106] gui_setup(): statusbar setup
[I][callback.cpp:92] callback_init(): init callback_t structure success for: bma
[I][callback.cpp:92] callback_init(): init callback_t structure success for: pmu
[I][callback.cpp:92] callback_init(): init callback_t structure success for: display
[I][gui.cpp:108] gui_setup(): quickbar setup
[I][gui.cpp:110] gui_setup(): keyboard setup
[I][gui.cpp:112] gui_setup(): num keyboard setup
[I][widget_factory.cpp:536] wf_get_settings_header_title(): header_title found
[I][basejsonconfig.cpp:135] load(): reading json failed, call defaults, file: /weather.json
[I][basejsonconfig.cpp:135] load(): reading json failed, call defaults, file: /sound.json
[I][basejsonconfig.cpp:123] load(): json config deserializeJson() success: Ok, file: /gpsctl.json
[I][basejsonconfig.cpp:354] save(): json config serializeJson() success: /gpsctl.json
[E][gpsctl.cpp:101] gpsctl_setup(): set default gps RX on pin -1/TX on pin -1!
[I][hardware.cpp:264] hardware_post_setup(): Total heap: 243016
[I][hardware.cpp:265] hardware_post_setup(): Free heap: 120876
[I][hardware.cpp:266] hardware_post_setup(): Total PSRAM: 0
[I][hardware.cpp:267] hardware_post_setup(): Free PSRAM: 0
[E][esp32-hal-misc.c:96] disableCore0WDT(): Failed to remove Core 0 IDLE task from WDT
`
please help me!

sharandac, 2022-01-11:
Thanks for your feedback and thanks for the hint. I will look at it later and try to fix it. The Watch-2021 unfortunately uses a controller that is not properly supported.

ILXR, 2022-01-11:
> Thanks for your feedback and thanks for the hint. I will look at it later and try to fix it. The Watch-2021 unfortunately uses a controller that is not properly supported.

ok, thanks a lot

sharandac, 2022-01-13:
It's probably a bit more complicated to fix. I will take a closer look at the weekend and try to fix it again.

ILXR, 2022-01-14:
> It's probably a bit more complicated to fix. I will take a closer look at the weekend and try to fix it again.

ok, I'm afraid it was a bother for you to do this.

commit 17ed57c3 (2022-06-30): fix #332 and enhance t-watch-2021 support
closed 2022-03-18 by sharandac