sharandac/My-TTGO-Watch #334: Fork/Amazfish Support

Crsarmv7l, 2022-01-12:
Hi Sharandc,

I appreciate all the help from both you and d03n3rfr1tz3 with regard to the random questions, bug fixes and help. 

About 5 minutes ago after months of searching I have finally squashed my "No Weather Message" when using a modified version of your firmware with the Amazfish companion app (available on linux phone distros and Sailfish OS). The issue was in the Amazfish code, and I have tracked it down and notified the developer as well as implemented a local fix on my phone.

I am not sure how many people would like to utilize that companion app, but I would be happy to provide you with my changes to integrate into the main project or I could create a fork so that the capability is available.

Thus far the following are working:
- Weather messages
- Email alerts
- Message alerts
- Calls
- Notifications
- Basics like time sync, clock etc

I probably will not implement too much more as those fit all my desires. Some of my modifications are pretty hacky. But please let me know if you would like to include it in your code.

sharandac, 2022-01-13:
Hey hey,

that sounds wonderful. I would suggest that you first create a fork, then we can look at it in peace and look for errors. In the meantime, we can also start thinking about how we can integrate your code. If that would be okay for you?

Crsarmv7l, 2022-01-16:
Sure, Like I said it is pretty hacky and parts may conflict with yours so if you want to just leave it as a fork that is fine too.

Once I finish enhancing the GB Weather message a bit I will get going on it.

closed 2022-01-16 by Crsarmv7l