sharandac/My-TTGO-Watch #337: Black Screen

xXDevboiXx, 2022-01-23:
Ok so i finshed the upload, sucessful and stuff, but heres the catch, my ttgo is not turning on.

xXDevboiXx, 2022-01-23:
> Ok so i finshed the upload, sucessful and stuff, but heres the catch, my ttgo is not turning on.

update, tried to upload hedge again but it got a time out waiting for the package to transmit, how do i switch on my watch in such a way it will transmit or recive the package (os wont work as the orignal os is wipedout.

Crsarmv7l, 2022-01-23:
.....I am not even going to speculate without an build log because based on #331 it could be any number of things

xXDevboiXx, 2022-01-24:
> .....I am not even going to speculate without a build log because based on #331 it could be any number of things

no , #331  was diffrent it was compiling error, but i reinstalled the driver and it worked, upload was sucessful, but the watch was not coming on, later when i tried uploading it again, the watch got timed out as the watch did not recive some packages

xXDevboiXx, 2022-01-24:
> .....I am not even going to speculate without a build log because based on #331 it could be any number of things

and yes, i am a child on github doing coding.

Crsarmv7l, 2022-01-24:
> no , #331 was diffrent it was compiling error, but i reinstalled the driver and it worked, upload was sucessful, but the watch was not coming on, later when i tried uploading it again, the watch got [...]

No. It wasn't a compiling error in #331, it was a you error setting up the driver.

In #331 you also indicated you don't know which version T-watch you have. Don't you think uploading the wrong version of code might have substantial issues?

None of these issues are at all related to this firmware or github.

So, good luck I guess? I have had a couple bootloops, but provided you didn't substantially mess up, simply initiating an upload from pio automatically puts it in download mode and overwrites what was on it. Again though that is assuming you didn't do something crazy like flashing the firmware for a completely different piece of hardware, which seems very likely.

xXDevboiXx, 2022-01-24:
yep, i do, T-WATCH-2020

xXDevboiXx, 2022-01-24:
but i dont have the tf card or mic, at least i think so

xXDevboiXx, 2022-01-24:
but i just have one question, how the hell will i switch on my twatch, with enough processing power to get the package

Crsarmv7l, 2022-01-24:
> but i dont have the tf card or mic, at least i think so

Again there are 3 different versions of the 2020 twatch. You need to know EXACTLY which one. Not "think."

>but i just have one question, how the hell will i switch on my twatch, with enough processing power to get the package

The watch doesn't need to turn on, again assuming you didn't flash the completely wrong firmware. You plug it in and tell platformio to upload.

You are on your own at this point

d03n3rfr1tz3, 2022-01-24:
If it does not react to anything and stays black, but should not be broken or empty, then it might just be turned off. Just press the button on the side, the only button this thing has, for about 5-10s. Also make sure it has enough power in its battery, which it should, if you connect it a few minutes via USB.

From there on you can continue were you stopped. Maybe the firmware was successfully flashed and it boots. Maybe it throws an error in the Serial Monitor. But whatever happens, try to be way more precious, copy over the serial output and/or build logs and also please use the edit-functionality here on github instead of creating new comments every few minutes.

xXDevboiXx, 2022-01-24:
> > but i dont have the tf card or mic, at least i think so
> 
> Again there are 3 different versions of the 2020 twatch and one 2021. You need to know EXACTLY which one. Not "think."
> 
> > but i just have one question, how the hell will i switch on my twatch, with enough processing power to get the package
> 
> The watch doesn't need to turn on, again assuming you didn't flash the completely wrong firmware. You plug it in and tell platformio to upload and it will.
> 
> You are on your own at this point

should i be screaming or smthn?

xXDevboiXx, 2022-01-24:
> If it does not react to anything and stays black, but should not be broken or empty, then it might just be turned off. Just press the button on the side, the only button this thing has, for about [...]

it does have power in its batteries, but the thing is i tried switching it on for about 20 secs or so, it did not work, so just to check if the ttgo watch has a bootloader mode, if so, how to access it

xXDevboiXx, 2022-01-25:
Yoooooo waassup fixed the watchhh

closed 2022-01-25 by xXDevboiXx