sharandac/My-TTGO-Watch #339: Keyboard

xXDevboiXx, 2022-01-25:
so the keyboard is nice but could you add a left arrow as well, its kinda hard going thru a bunchhh of panels until finally reach the panel you want to reach

Crsarmv7l, 2022-01-25:
I am glad you were able to get your watch working. 

That is the nice part of open source software, you can take this software and modify it to function how you want.

xXDevboiXx, 2022-01-25:
> I am glad you were able to get your watch working.
> 
> That is the nice part of open source software, you can take this software and modify it to function how you want. Plus you learn quite a bit. I am a novice at best, but after 6+ months of messing with this software and modifying it I have certainly learned quite a bit and have a neat watch working with an obscure OS as well.
> 
> Have fun.

Thank you! And yes i will defintly have fun with it but i really hope the keyboard qill become better also (for the devs) can you add a web browser and download manager or smthn would defintly make it fun as hell

closed 2022-01-25 by xXDevboiXx