sharandac/My-TTGO-Watch #357: Help with first app.. SSL causing reboot?

asdf23, 2022-04-22:
Greetings & awesome job btw. 
I've been trying to create a JIRA app. Jira is a software development tracking system.  They have a REST API that requires https. I don't know if that is a deal breaker. 

I copied much of fx_rates.cpp and almost have a working app. The REST APIi requires a basic authentication header so I added that to uri_load.cpp (copied uri_load_http_to_ram, uri_load_https_to_ram, and JsonRequest::process creating uri_load_http_to_ram_with_basic_auth, uri_load_https_to_ram_with_basic_auth, and JsonRequest::processWithBasicAuth)

(Happy to upload all of this, but not sure if it's valuable at this point in it's broken state)

Anyway the app calls the api, and then reboots the phone. It seems to reboot right after start_ssl_client(). The last log is "[I][ssl_client.cpp:127] start_ssl_client(): WARNING: Skipping SSL Verification. INSECURE!"

Any ideas what I could be doing wrong or if this is possible?

The entire log is:

```
[I][syncapp.cpp:123] SyncTaskHandler(): SyncTaskHandler: 1073510088
[I][syncapp.cpp:98] onSyncRequest(): start jira tickets sync Task, heap: 77404
[I][uri_load.cpp:641] uri_load_http_to_ram_with_basic_auth(): in uri_load_http_to_ram_with_basic_auth http://myCloudSite.atlassian.net/rest/api/3/search?jql=filter=40849&fields=summary XXXXXXXXX
[I][uri_load.cpp:730] uri_load_http_to_ram_with_basic_auth(): 301/302 redirect to: https://myCloudSite.atlassian.net/rest/api/3/search?jql=filter=40849&fields=summary
[E][uri_load.cpp:743] uri_load_http_to_ram_with_basic_auth(): location2 c_str
[I][uri_load.cpp:1091] uri_load_https_to_ram_with_basic_auth(): in uri_load_https_to_ram_with_basic_auth https://myCloudSite.atlassian.net/rest/api/3/search?jql=filter=40849&fields=summary XXXXXXXXX
[I][ssl_client.cpp:127] start_ssl_client(): WARNING: Skipping SSL Verification. INSECURE!
Guru Meditation Error: Core  1 panic'ed (Unhandled debug exception)
Debug exception reason: Stack canary watchpoint triggered (jira tickets sy) 
Core 1 register dump:
PC      : 0x40097318  PS      : 0x00060d36  A0      : 0x4008a985  A1      : 0x3fff10d0  
A2      : 0x2fc9a2fd  A3      : 0xb6e1b300  A4      : 0xddd38313  A5      : 0x00000038  
A6      : 0x49ca978e  A7      : 0x0c23e695  A8      : 0xbf212e1e  A9      : 0xf504fc11  
A10     : 0xb6e1b300  A11     : 0xaf440fcb  A12     : 0x49ca978e  A13     : 0xb5dafc97  
A14     : 0x7783bc57  A15     : 0x35406c71  SAR     : 0x00000008  EXCCAUSE: 0x00000001  
EXCVADDR: 0x00000000  LBEG    : 0x4000c46c  LEND    : 0x4000c477  LCOUNT  : 0xffffffff  

ELF file SHA256: 0000000000000000

Backtrace: 0x40097318:0x3fff10d0 0x4008a982:0x3fff14e0 0x401e97c1:0x3fff1500 0x401e9492:0x3fff1520 0x401e8332:0x3fff1590 0x401e84cf:0x3fff1610 0x401e2f61:0x3fff1630 0x401e4c9f:0x3fff16a0 0x401e51e9:0x3fff1760 0x401e526b:0x3fff1790 0x401e529f:0x3fff17c0 0x401f680b:0x3fff1810 0x401f6a51:0x3fff18b0 0x401f6a8a:0x3fff1900 0x401eab4d:0x3fff1920 0x401eab87:0x3fff1940 0x401e98af:0x3fff1a10 0x401f9d40:0x3fff1a30 0x401fa49b:0x3fff1ac0 0x401f18fd:0x3fff1ae0 0x401f193d:0x3fff1b00 0x401ab05d:0x3fff1b20 0x401aa876:0x3fff1da0 0x401aa9c9:0x3fff1dd0 0x4030ea1b:0x3fff1df0 0x401ac65e:0x3fff1e10 0x401ac6ef:0x3fff1e40 0x401ac86b:0x3fff1e90 0x4015c303:0x3fff1eb0 0x4015bae1:0x3fff2050 0x4015bfaf:0x3fff2070 0x4015bac6:0x3fff2210 0x4015bb65:0x3fff2230 0x4010cd10:0x3fff2250 0x400dc016:0x3fff2280 0x400dc167:0x3fff2380 0x4010c0e6:0x3fff23b0 0x4010c062:0x3fff23e0 0x40094266:0x3fff2400
  #0  0x40097318:0x3fff10d0 in _frxt_int_enter at /home/sharan/temp/esp32-arduino-lib-builder/esp-idf/components/freertos/portasm.S:105
  #1  0x4008a982:0x3fff14e0 in _xt_lowint1 at /home/sharan/temp/esp32-arduino-lib-builder/esp-idf/components/freertos/xtensa_vectors.S:1139
  #2  0x401e97c1:0x3fff1500 in sha384_update_wrap at /home/sharan/temp/esp32-arduino-lib-builder/esp-idf/components/mbedtls/mbedtls/library/md_wrap.c:526
  #3  0x401e9492:0x3fff1520 in mbedtls_md_hmac_finish at /home/sharan/temp/esp32-arduino-lib-builder/esp-idf/components/mbedtls/mbedtls/library/md.c:161
  #4  0x401e8332:0x3fff1590 in mbedtls_hmac_drbg_update_ret at /home/sharan/temp/esp32-arduino-lib-builder/esp-idf/components/mbedtls/mbedtls/library/hmac_drbg.c:348
  #5  0x401e84cf:0x3fff1610 in mbedtls_hmac_drbg_seed_buf at /home/sharan/temp/esp32-arduino-lib-builder/esp-idf/components/mbedtls/mbedtls/library/hmac_drbg.c:348
  #6  0x401e2f61:0x3fff1630 in ecp_drbg_seed at /home/sharan/temp/esp32-arduino-lib-builder/esp-idf/components/mbedtls/mbedtls/library/ecp.c:3197
  #7  0x401e4c9f:0x3fff16a0 in ecp_mul_comb at /home/sharan/temp/esp32-arduino-lib-builder/esp-idf/components/mbedtls/mbedtls/library/ecp.c:3197
  #8  0x401e51e9:0x3fff1760 in mbedtls_ecp_mul_restartable at /home/sharan/temp/esp32-arduino-lib-builder/esp-idf/components/mbedtls/mbedtls/library/ecp.c:3197
  #9  0x401e526b:0x3fff1790 in mbedtls_ecp_mul_shortcuts at /home/sharan/temp/esp32-arduino-lib-builder/esp-idf/components/mbedtls/mbedtls/library/ecp.c:3197
  #10 0x401e529f:0x3fff17c0 in mbedtls_ecp_muladd_restartable at /home/sharan/temp/esp32-arduino-lib-builder/esp-idf/components/mbedtls/mbedtls/library/ecp.c:3197
  #11 0x401f680b:0x3fff1810 in ecdsa_verify_restartable at /home/sharan/temp/esp32-arduino-lib-builder/esp-idf/components/mbedtls/mbedtls/library/ecdsa.c:932
  #12 0x401f6a51:0x3fff18b0 in mbedtls_ecdsa_read_signature_restartable at /home/sharan/temp/esp32-arduino-lib-builder/esp-idf/components/mbedtls/mbedtls/library/ecdsa.c:932
  #13 0x401f6a8a:0x3fff1900 in mbedtls_ecdsa_read_signature at /home/sharan/temp/esp32-arduino-lib-builder/esp-idf/components/mbedtls/mbedtls/library/ecdsa.c:932
  #14 0x401eab4d:0x3fff1920 in ecdsa_verify_wrap at /home/sharan/temp/esp32-arduino-lib-builder/esp-idf/components/mbedtls/mbedtls/library/pk_wrap.c:509
  #15 0x401eab87:0x3fff1940 in eckey_verify_wrap at /home/sharan/temp/esp32-arduino-lib-builder/esp-idf/components/mbedtls/mbedtls/library/pk_wrap.c:274
  #16 0x401e98af:0x3fff1a10 in mbedtls_pk_verify_restartable at /home/sharan/temp/esp32-arduino-lib-builder/esp-idf/components/mbedtls/mbedtls/library/pk.c:560
  #17 0x401f9d40:0x3fff1a30 in ssl_parse_server_key_exchange at /home/sharan/temp/esp32-arduino-lib-builder/esp-idf/components/mbedtls/mbedtls/library/ssl_cli.c:2904
  #18 0x401fa49b:0x3fff1ac0 in mbedtls_ssl_handshake_client_step at /home/sharan/temp/esp32-arduino-lib-builder/esp-idf/components/mbedtls/mbedtls/library/ssl_cli.c:3840
  #19 0x401f18fd:0x3fff1ae0 in mbedtls_ssl_handshake_step at /home/sharan/temp/esp32-arduino-lib-builder/esp-idf/components/mbedtls/mbedtls/library/ssl_tls.c:7875
  #20 0x401f193d:0x3fff1b00 in mbedtls_ssl_handshake at /home/sharan/temp/esp32-arduino-lib-builder/esp-idf/components/mbedtls/mbedtls/library/ssl_tls.c:7875
  #21 0x401ab05d:0x3fff1b20 in start_ssl_client(sslclient_context*, char const*, unsigned int, int, char const*, char const*, char const*, char const*, char const*, bool) at /home/user1/.platformio/packages/framework-arduinoespressif32/libraries/WiFiClientSecure/src/ssl_client.cpp:411
  #22 0x401aa876:0x3fff1da0 in WiFiClientSecure::connect(char const*, unsigned short, char const*, char const*, char const*) at /home/user1/.platformio/packages/framework-arduinoespressif32/libraries/WiFiClientSecure/src/WiFiClientSecure.cpp:344
  #23 0x401aa9c9:0x3fff1dd0 in WiFiClientSecure::connect(char const*, unsigned short) at /home/user1/.platformio/packages/framework-arduinoespressif32/libraries/WiFiClientSecure/src/WiFiClientSecure.cpp:344
  #24 0x4030ea1b:0x3fff1df0 in WiFiClientSecure::connect(char const*, unsigned short, int) at /home/user1/.platformio/packages/framework-arduinoespressif32/libraries/WiFiClientSecure/src/WiFiClientSecure.cpp:117
  #25 0x401ac65e:0x3fff1e10 in HTTPClient::connect() at /home/user1/.platformio/packages/framework-arduinoespressif32/libraries/HTTPClient/src/HTTPClient.cpp:1460
  #26 0x401ac6ef:0x3fff1e40 in HTTPClient::sendRequest(char const*, unsigned char*, unsigned int) at /home/user1/.platformio/packages/framework-arduinoespressif32/libraries/HTTPClient/src/HTTPClient.cpp:1460
  #27 0x401ac86b:0x3fff1e90 in HTTPClient::GET() at /home/user1/.platformio/packages/framework-arduinoespressif32/libraries/HTTPClient/src/HTTPClient.cpp:1460
  #28 0x4015c303:0x3fff1eb0 in uri_load_https_to_ram_with_basic_auth(uri_load_dsc_t*, char const*) at src/utils/uri_load/uri_load.cpp:355
  #29 0x4015bae1:0x3fff2050 in uri_load_to_ram_with_basic_auth(char const*, char const*, void (*)(int)) at src/utils/uri_load/uri_load.cpp:355
[... cut ...]
  #36 0x4010c0e6:0x3fff23b0 in std::function<void (SyncRequestSource)>::operator()(SyncRequestSource) const at /home/user1/.platformio/packages/toolchain-xtensa32/xtensa-esp32-elf/include/c++/5.2.0/functional:2271
      (inlined by) SynchronizedApplication::onSyncRequest() at src/quickglui/app/syncapp.cpp:107
  #37 0x4010c062:0x3fff23e0 in SynchronizedApplication::SyncTaskHandler(void*) at src/quickglui/app/syncapp.cpp:125
  #38 0x40094266:0x3fff2400 in vPortTaskWrapper at /home/sharan/temp/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c:355 (discriminator 1)

Rebooting...
```

asdf23, 2022-04-22:
Lines 28->33 are my changes to stick the basic auth header in there. I struggled with this a bit, but I don' really think it's related to this problem...?

Lines 34-35 are the app I created. It's passing the URL and Basic Auth value down stream. Otherwise it's an exact copy of fx_rates.  Later I hope to add the textbox widget from the calendar and make a scrollable list of tickets.

Help!

sharandac, 2022-04-23:
Thanks for Your Feedback! 

It look like fullAuthValue is too short and strcpy and strcat is write out there and destroys your stack. Please use strlcpy and strlcat to prevent this.

```c
    const char* authHeaderStart = "Basic ";
    char fullAuthValue[ strlen( auth_string ) + strlen( authHeaderStart ) + sizeof( char ) ];
    strlcpy( fullAuthValue, authHeaderStart, sizeof( fullAuthValue ) );
    strlcat( fullAuthValue, auth_string, , sizeof( fullAuthValue ) );
```

Or on the the other side, the stack is to small for a recursive action. How big is your stack in this task? Both can trigger "Debug exception reason: Stack canary watchpoint triggered".

asdf23, 2022-04-24:
Thank you for your response! It makes me feel dumb :) I haven't programmed in C++ in a number of decades and was never very good at it. I'll give it a shot and let you know!

asdf23, 2022-04-24:
Well you're absolutely right. I never thought to doubt this part of the code. Now I have hard coded all of the strings.  I past this error but still didn't not complete an API call.

I am sorry I did not understand your question "How big is your stack in this task? ".   The response of the JSON should be a few characters long perhaps less than 3,000 characters. The program itself is a poor clone of fx_rates.cpp   

load_uri.cpp is now
```
...snip...
    download_client.addHeader("Authorization", "XXXXXX");
...snip...
````

It's still crashing/rebooting but differently.  I wonder if I need to provide a certificate?  I found this article, not sure if my problem is related: https://techtutorialsx.com/2017/11/18/esp32-arduino-https-get-request/ 

```
[I][timesync.cpp:324] timesync_Task(): finish time sync task, heap: 82924
[I][update.cpp:329] update_Task(): star[I][syncappt u.date task:1heapS 77T53k
Handle]()ri_lnadaspH:433] urier: 107_load_htt3_to088
p: 72148uri[Ioad_ncappto_ram h.cpp:98] ontSyncRequest()tp://www.neo-gutartlazra /ttgo-ttickets sync-watch20k, h3.version.json
[I][uri_load.cpp:1115] uri_load_https_to_ram_with_basic_auth(): in uri_load_https_to_ram_with_basic_auth https://vitacost.atlassian.net/rest/api/3/search?jql=filter=40849&fields=summary YW5kcmV3Lm1haW5sYW5kQGtyb2dlci5jb206OTU4RTNlbjgwY3pRRzVTZzRuM1czREJC
[I][ssl_client.cpp:127] start_ssl_client(): WARNING: Skipping SSL Verification. INSECURE!
[I][update_check_version.cpp:46] update_check_new_version(): load update information from: http://www.neo-guerillaz.de/ttgo-t-watch2020_v3.version.json
[I][update.cpp:396] update_Task(): finish update task, heap: 31548
Guru Meditation Error: Core  1 panic'ed (Unhandled debug exception)
Debug exception reason: Stack canary watchpoint triggered (jira tickets sy) 
Core 1 register dump:
PC      : 0x4008a969  PS      : 0x00060736  A0      : 0x00060730  A1      : 0x3fff1b70  
A2      : 0xbc6f1c0d  A3      : 0x565ce9aa  A4      : 0x08ad8cd1  A5      : 0x84a391f9  
A6      : 0x00000027  A7      : 0xce9f0708  A8      : 0x3fff1c60  A9      : 0xfb36c85a  
A10     : 0x3fff1c60  A11     : 0x3fff1c60  A12     : 0x3fff1c60  A13     : 0x1f83d9ab  
A14     : 0xdf13dfbc  A15     : 0x36231eae  SAR     : 0x00000018  EXCCAUSE: 0x00000001  
EXCVADDR: 0x00000000  LBEG    : 0x4000c46c  LEND    : 0x4000c477  LCOUNT  : 0x00000000  

ELF file SHA256: 0000000000000000

Backtrace: 0x4008a969:0x3fff1b70 0x0006072d:0x3fff1f80 0x401e96e1:0x3fff1fa0 0x401e93e3:0x3fff1fc0 0x401e8266:0x3fff2030 0x401e84e2:0x3fff20b0 0x401e8525:0x3fff20e0 0x401e2e59:0x3fff2100 0x401e1f7a:0x3fff2120 0x401e342d:0x3fff2150 0x401e4a9e:0x3fff21a0 0x401e4b54:0x3fff2200 0x401e4cd9:0x3fff2350 0x401e511d:0x3fff2410 0x401e519f:0x3fff2440 0x401e51d3:0x3fff2470 0x401f673f:0x3fff24c0 0x401f6985:0x3fff2560 0x401f69be:0x3fff25b0 0x401eaa81:0x3fff25d0 0x401eaabb:0x3fff25f0 0x401e97e3:0x3fff26c0 0x401f9c74:0x3fff26e0 0x401fa3cf:0x3fff2770 0x401f1831:0x3fff2790 0x401f1871:0x3fff27b0 0x401aaf91:0x3fff27d0 0x401aa7aa:0x3fff2a50 0x401aa8fd:0x3fff2a80 0x4030e94f:0x3fff2aa0 0x401ac592:0x3fff2ac0 0x401ac623:0x3fff2af0 0x401ac79f:0x3fff2b40 0x4015c242:0x3fff2b60 0x4015ba6d:0x3fff2cb0 0x4015baf1:0x3fff2cd0 0x4010cc9c:0x3fff2cf0 0x400dbfa2:0x3fff2d20 0x400dc0f3:0x3fff2dd0 0x4010c072:0x3fff2e00 0x4010bfee:0x3fff2e30 0x40094266:0x3fff2e50
  #0  0x4008a969:0x3fff1b70 in _xt_lowint1 at /home/sharan/temp/esp32-arduino-lib-builder/esp-idf/components/freertos/xtensa_vectors.S:1120
  #1  0x0006072d:0x3fff1f80 in ?? ??:0
  #2  0x401e96e1:0x3fff1fa0 in sha384_finish_wrap at /home/sharan/temp/esp32-arduino-lib-builder/esp-idf/components/mbedtls/mbedtls/library/md_wrap.c:532
  #3  0x401e93e3:0x3fff1fc0 in mbedtls_md_hmac_finish at /home/sharan/temp/esp32-arduino-lib-builder/esp-idf/components/mbedtls/mbedtls/library/md.c:161
  #4  0x401e8266:0x3fff2030 in mbedtls_hmac_drbg_update_ret at /home/sharan/temp/esp32-arduino-lib-builder/esp-idf/components/mbedtls/mbedtls/library/hmac_drbg.c:348
  #5  0x401e84e2:0x3fff20b0 in mbedtls_hmac_drbg_random_with_add at /home/sharan/temp/esp32-arduino-lib-builder/esp-idf/components/mbedtls/mbedtls/library/hmac_drbg.c:411
  #6  0x401e8525:0x3fff20e0 in mbedtls_hmac_drbg_random at /home/sharan/temp/esp32-arduino-lib-builder/esp-idf/components/mbedtls/mbedtls/library/hmac_drbg.c:436
  #7  0x401e2e59:0x3fff2100 in ecp_drbg_random at /home/sharan/temp/esp32-arduino-lib-builder/esp-idf/components/mbedtls/mbedtls/library/ecp.c:3197
  #8  0x401e1f7a:0x3fff2120 in mbedtls_mpi_fill_random at /home/sharan/temp/esp32-arduino-lib-builder/esp-idf/components/mbedtls/mbedtls/library/bignum.c:2351
  #9  0x401e342d:0x3fff2150 in ecp_randomize_jac at /home/sharan/temp/esp32-arduino-lib-builder/esp-idf/components/mbedtls/mbedtls/library/ecp.c:3197
  #10 0x401e4a9e:0x3fff21a0 in ecp_mul_comb_core at /home/sharan/temp/esp32-arduino-lib-builder/esp-idf/components/mbedtls/mbedtls/library/ecp.c:3197
  #11 0x401e4b54:0x3fff2200 in ecp_mul_comb_after_precomp at /home/sharan/temp/esp32-arduino-lib-builder/esp-idf/components/mbedtls/mbedtls/library/ecp.c:3197
  #12 0x401e4cd9:0x3fff2350 in ecp_mul_comb at /home/sharan/temp/esp32-arduino-lib-builder/esp-idf/components/mbedtls/mbedtls/library/ecp.c:3197
  #13 0x401e511d:0x3fff2410 in mbedtls_ecp_mul_restartable at /home/sharan/temp/esp32-arduino-lib-builder/esp-idf/components/mbedtls/mbedtls/library/ecp.c:3197
[... cut ...]
  #39 0x4010c072:0x3fff2e00 in std::function<void (SyncRequestSource)>::operator()(SyncRequestSource) const at /home/user1/.platformio/packages/toolchain-xtensa32/xtensa-esp32-elf/include/c++/5.2.0/functional:2271
      (inlined by) SynchronizedApplication::onSyncRequest() at src/quickglui/app/syncapp.cpp:107
  #40 0x4010bfee:0x3fff2e30 in SynchronizedApplication::SyncTaskHandler(void*) at src/quickglui/app/syncapp.cpp:125
  #41 0x40094266:0x3fff2e50 in vPortTaskWrapper at /home/sharan/temp/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c:355 (discriminator 1)

Rebooting...

```

Any ideas?

asdf23, 2022-04-25:
Seems like this is the issue: https://community.platformio.org/t/esp32-stack-configuration-reloaded/20994/2

... App crashes when it is performing a ... SSL... command with the debugger message: ‘ERROR A stack overflow in task loopTask has been detected’. The stack overflow seems to happen somewhere in the mbedtls sha512 calculations.

Increasing CONFIG_ARDUINO_LOOP_STACK_SIZE .... to 16384

I did try this it did not work for me :(

sharandac, 2022-04-25:
Stay calm! What is noticeable in the log file is that the memory for programmes, i.e. the heap, is running out.

```
[I][update.cpp:396] update_Task(): finish update task, heap: 31548
```

Disable the autosync of all programmes (weather, update, timesync and so on), so that not all programmes start when Wifi is connected. SSL needs at least 80k free heap.

asdf23, 2022-04-25:
Rather than take you're calming advise :)  I set the STACK_SIZE to  `#define CONFIG_ARDUINO_LOOP_STACK_SIZE 32768`
( `~/platform.io/packages/framework-arduinoespressif32/tools/sdk/include/config/sdkconfig.h` btw )

This sort of worked ~~ the reboot is not occurring and instead I am getting this:

```
[I][syncapp.cpp:123] SyncTaskHandler(): SyncTaskHandler: 1073510088
[I][syncapp.cpp:98] onSyncRequest(): start jira tickets sync Task, heap: 53428
[I][uri_load.cpp:1115] uri_load_https_to_ram_with_basic_auth(): in uri_load_https_to_ram_with_basic_auth https://myCloud.atlassian.net/rest/api/3/search?jql=filter=40849&fields=summary 
[E][WiFiGeneric.cpp:738] hostByName(): DNS Failed for myCloud.atlassian.net
[E][WiFiClientSecure.cpp:133] connect(): start_ssl_client: -1
[W][HTTPClient.cpp:1404] returnError(): error(-1): connection refused
[E][uri_load.cpp:188] uri_load_to_ram_with_basic_auth(): failed uri ..             <----- BTW
[E][uri_load.cpp:189] uri_load_to_ram_with_basic_auth(): uri not supported..
[E][uri_load.cpp:1265] uri_load_https_to_ram_with_basic_auth(): redirect failed
[I][syncapp.cpp:111] onSyncRequest(): finsh jira tickets sync Task, heap: 53428
[I][pmu.cpp:374] pmu_loop(): AXP202: TimerTimeoutIRQ
[I][powermgm.cpp:114] powermgm_loop(): go silence wakeup
```

So it sort of works minus DNS not working. 

That logging statement is coded like this: 

```
        /**
         * check for uri file source
         */
        if ( strstr( uri, "http://" ) ) {
            URI_LOAD_LOG("http source");
            uri_load_dsc = uri_load_http_to_ram_with_basic_auth( uri_load_dsc, auth_string );
        }
        else if ( strstr( uri, "https://" ) ) {
            URI_LOAD_LOG("https source");
            uri_load_dsc = uri_load_https_to_ram_with_basic_auth( uri_load_dsc, auth_string );
        }
        else if ( strstr( uri, "file://" ) ) {
            URI_LOAD_LOG("local files source");
            uri_load_dsc = uri_load_file_to_ram( uri_load_dsc );
        }
        else {
            URI_LOAD_ERROR_LOG("failed uri %s..", uri);       <----------- HERE
            URI_LOAD_ERROR_LOG("uri not supported..");
            uri_load_free_all( uri_load_dsc );
            uri_load_dsc = NULL;
        }
```

I believe this might be a bug. If DNS fails the URL becomes `""`  but the request continues.

sharandac, 2022-04-25:
Ah ... the error come from another line of code. When we have no HTTP_OK from the last http request, i clear the location string and check if we have a 301 or 302 http code ( new location or redirect ) and full the location string with the new location.
After that i check the location string and start a new http request.

a simple change from

```c
if ( location ) {
```
to
```c
if ( location.isEmpty() ) {
```
can solve this problem. I fix this in the next version.

asdf23, 2022-04-26:
Thank you for your help with this. I'm going to close this and wish the next developer luck with it.

In summary make a very large stack (to prevent the canary issue), browse to the site on the connected router before connecting the watch (to clear the DNS issue). After this I still experienced SSL issues. Possibly there is an answer to this problem here: https://techtutorialsx.com/2017/11/18/esp32-arduino-https-get-request/ 

My plan was to collect the jira site,  filter id, API user key from the config menu. Use this to get a list of tickets and titles over a json REST API then display this in N scrollable textboxes as found in the calendar app. 

I believe that the ESP32 device just isn't up to the challenge.

closed 2022-04-26 by asdf23