sharandac/My-TTGO-Watch #358: Compile success , upload failed with A fatal error occurred: Invalid head of packet (0x5C)

bentech4u, 2022-04-24:
HI,

I am very new to T-Watch. I have installed platform.io(vscode)

download watch repository , build was success , while uploading it got failed.

```
Auto-detected: /dev/ttyACM0
Uploading .pio/build/t-watch2020-v3/firmware.bin
esptool.py v3.1
Serial port /dev/ttyACM0
Connecting.....
Chip is ESP32-D0WDQ6-V3 (revision 3)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: 58:bf:25:05:d4:dc
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 2000000
Changed.
Configuring flash size...
Auto-detected Flash size: 16MB
Flash will be erased from 0x00001000 to 0x00005fff...
Flash will be erased from 0x00008000 to 0x00008fff...
Flash will be erased from 0x0000e000 to 0x0000ffff...
Flash will be erased from 0x00010000 to 0x00556fff...
Compressed 18544 bytes to 12051...
Writing at 0x00001000... (100 %)
Wrote 18544 bytes (12051 compressed) at 0x00001000 in 0.4 seconds (effective 422.2 kbit/s)...
Hash of data verified.
Compressed 3072 bytes to 129...
Writing at 0x00008000... (100 %)
Wrote 3072 bytes (129 compressed) at 0x00008000 in 0.1 seconds (effective 453.5 kbit/s)...
Hash of data verified.
Compressed 8192 bytes to 47...
Writing at 0x0000e000... (100 %)
Wrote 8192 bytes (47 compressed) at 0x0000e000 in 0.1 seconds (effective 605.8 kbit/s)...
Hash of data verified.
Compressed 5532672 bytes to 2555448...
Writing at 0x00010000... (0 %)
Writing at 0x0001710d... (1 %)
Writing at 0x00044371... (1 %)
Writing at 0x0004b2a9... (2 %)
Writing at 0x000b27a6... (3 %)

A fatal error occurred: Invalid head of packet (0x5C)
*** [upload] Error 2
========================================================================= [FAILED] Took 24.07 seconds =========================================================================

Environment     Status    Duration
--------------  --------  ------------
t-watch2020-v3  FAILED    00:00:24.066
```

sharandac, 2022-04-25:
Thanks for your feedback! It looks like some USB driver issues and not an error from PIO itself. Please check them and try again.

bentech4u, 2022-04-26:
moreover when i tried from Arduino IDE, with example codes. it is writing to watch, that time it uses same usb driver

sharandac, 2022-04-26:
Thanks for your feedback. Another problem could perhaps be the programming speed. Not all interfaces manage the default speeds. Insert the following line in the appropriate environment ( [env:t-watch2020-v3] ). What speed does the Arduino IDE use for you?

```
upload_speed = 921600
```

bentech4u, 2022-04-26:
One of the successful upload from Arduino IDE as bellow:

```
esptool.py v3.3-dev
Serial port /dev/ttyACM0
Connecting....
Chip is ESP32-D0WDQ6-V3 (revision 3)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: 58:bf:25:05:d4:dc
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 2000000
Changed.
Configuring flash size...
Flash will be erased from 0x0000e000 to 0x0000ffff...
Flash will be erased from 0x00001000 to 0x00005fff...
Flash will be erased from 0x00010000 to 0x000aefff...
Flash will be erased from 0x00008000 to 0x00008fff...
Compressed 8192 bytes to 47...
Writing at 0x0000e000... (100 %)
Wrote 8192 bytes (47 compressed) at 0x0000e000 in 0.1 seconds (effective 616.6 kbit/s)...
Hash of data verified.
Flash params set to 0x024f
Compressed 17120 bytes to 11841...
Writing at 0x00001000... (100 %)
Wrote 17120 bytes (11841 compressed) at 0x00001000 in 0.4 seconds (effective 379.9 kbit/s)...
Hash of data verified.
Compressed 648400 bytes to 317370...
Writing at 0x00010000... (5 %)
Writing at 0x000277ad... (10 %)
Writing at 0x00031abc... (15 %)
Writing at 0x0003a7d8... (20 %)
Writing at 0x00041337... (25 %)
Writing at 0x0004a96c... (30 %)
Writing at 0x0005c935... (35 %)
Writing at 0x000620f2... (40 %)
Writing at 0x00067e34... (45 %)
Writing at 0x0006d939... (50 %)
Writing at 0x00072d18... (55 %)
Writing at 0x0007824e... (60 %)
Writing at 0x0007e99e... (65 %)
Writing at 0x0008431d... (70 %)
Writing at 0x00089d33... (75 %)
Writing at 0x0008f550... (80 %)
Writing at 0x000985f5... (85 %)
Writing at 0x000a0e70... (90 %)
Writing at 0x000a6a80... (95 %)
Writing at 0x000ac46d... (100 %)
Wrote 648400 bytes (317370 compressed) at 0x00010000 in 5.8 seconds (effective 895.4 kbit/s)...
Hash of data verified.
Compressed 3072 bytes to 129...
Writing at 0x00008000... (100 %)
Wrote 3072 bytes (129 compressed) at 0x00008000 in 0.1 seconds (effective 464.0 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...

```

sharandac, 2022-04-26:
Ok. The only difference now is the size of the programme that has to be stored in the flash memory. What hardware are you using exactly?

bentech4u, 2022-04-26:
T-Watch V3

sharandac, 2022-04-26:
Then I'm now running out of ideas ...

bentech4u, 2022-04-28:
Any help,

Here is detailed UDB driver :

```
ben@bentech:/dev$ lsusb | grep QinHeng
Bus 001 Device 015: ID 1a86:55d4 QinHeng Electronics USB Single Serial
ben@bentech:/dev$ udevadm info --query all --name /dev/ttyACM0
P: /devices/pci0000:00/0000:00:14.0/usb1/1-1/1-1:1.0/tty/ttyACM0
N: ttyACM0
L: 0
S: serial/by-id/usb-1a86_USB_Single_Serial_537D022405-if00
S: serial/by-path/pci-0000:00:14.0-usb-0:1:1.0
E: DEVPATH=/devices/pci0000:00/0000:00:14.0/usb1/1-1/1-1:1.0/tty/ttyACM0
E: DEVNAME=/dev/ttyACM0
E: MAJOR=166
E: MINOR=0
E: SUBSYSTEM=tty
E: USEC_INITIALIZED=15221768079
E: ID_BUS=usb
E: ID_VENDOR_ID=1a86
E: ID_MODEL_ID=55d4
E: ID_PCI_CLASS_FROM_DATABASE=Serial bus controller
E: ID_PCI_SUBCLASS_FROM_DATABASE=USB controller
E: ID_PCI_INTERFACE_FROM_DATABASE=XHCI
E: ID_VENDOR_FROM_DATABASE=QinHeng Electronics
E: ID_VENDOR=1a86
E: ID_VENDOR_ENC=1a86
E: ID_MODEL=USB_Single_Serial
E: ID_MODEL_ENC=USB\x20Single\x20Serial
E: ID_REVISION=0443
E: ID_SERIAL=1a86_USB_Single_Serial_537D022405
E: ID_SERIAL_SHORT=537D022405
E: ID_TYPE=generic
E: ID_USB_INTERFACES=:020201:0a0000:
E: ID_USB_INTERFACE_NUM=00
E: ID_USB_DRIVER=cdc_acm
E: ID_USB_CLASS_FROM_DATABASE=Communications
E: ID_PATH=pci-0000:00:14.0-usb-0:1:1.0
E: ID_PATH_TAG=pci-0000_00_14_0-usb-0_1_1_0
E: ID_MM_CANDIDATE=1
E: ID_FOR_SEAT=tty-pci-0000_00_14_0-usb-0_1_1_0
E: DEVLINKS=/dev/serial/by-id/usb-1a86_USB_Single_Serial_537D022405-if00 /dev/serial/by-path/pci-0000:00:14.0-usb-0:1:1.0
E: TAGS=:seat:systemd:uaccess:

```

closed 2022-05-06 by sharandac