sharandac/My-TTGO-Watch #361: Abnormal after WiFi goes off screen

lewisxhe, 2022-05-31:
https://github.com/sharandac/My-TTGO-Watch/blob/1b085c0e011aab5a7dc416111d2a1b43d1c87931/src/hardware/wifictl.cpp#L649

`esp_wifi_stop();` should not be called here, it will cause the wifi to turn off after turning off the screen, just call `WiFi.mode( WIFI_OFF ); `

sharandac, 2022-05-31:
Thanks for your feedback. esp_wifi_stop() and esp_wifi_start() are intended to reset the driver control structure. For example, if promiscuous mode has been activated by an APP. Wifi.mode() does not do this. 
Can you describe exactly the effect that occurs?

lewisxhe, 2022-06-09:
In V1, V2, V3, if you turn off the screen and then turn on the wifi scan, it will not scan any device, then restart it, you can scan the device for the first time, because it has not called esp_wifi_stop, when the screen is turned off, call esp_wifi_stop , turning on the screen again will not find any devices, I think you should call WiFi.mode(WIFI_OFF) instead of esp_wifi_stop, everything works fine after I comment out esp_wifi_stop. Of course I just tested the wifi scan, I also noticed that Some places also call esp_wifi_stop

commit 30e0480b (2022-06-09): fix #361