sharandac/My-TTGO-Watch #366: TWatch 2021

playmiel, 2022-07-25:
- framework-arduinoespressif32 @ 2.0.4+sha.c5c5909
 - tool-esptoolpy @ 1.30300.0 (3.3.0)
 - tool-mkfatfs @ 2.0.1
 - tool-mklittlefs @ 1.203.210628 (2.3)
 - tool-mkspiffs @ 2.230.0 (2.30)
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch3

- [ ] TWatch 2021

round app not work in twatch2021 the edges of the screen app are unusable and keyboard it to big

it is possible to reduce the size of each application?

sharandac, 2022-07-26:
Thank you for your feedback. Unfortunately, most apps are not yet adapted for round displays. I have only managed to get the ESP32 working with PSRAM support. I think in the next few weeks I will have managed to work my way through it.
Some apps already have support via the three buttons.

[image]

Maybe I will get some support.

playmiel, 2022-07-26:
Ah very well I thought that everything was already adapted to the t-watch2021 in this case I will work on my side to adapt each application, I will make a commit if it works well

sharandac, 2022-07-26:
```c
void foo_tile_setup( uint32_t foo_tile_num ) {
    .
    .
    mainbar_add_tile_button_cb( foo_tile_num, foo_button_event_cb );
}

static bool foo_button_event_cb( EventBits_t event, void *arg ) {
    switch( event ) {
        case BUTTON_EXIT:           mainbar_jump_back();
                                    break;
        case BUTTON_SETUP:          foo_jump_to_setup();
                                    break;
        case BUTTON_REFRESH:        foo_refresh();
                                    break;
    }
    return( true );
}
```

closed 2022-07-26 by sharandac