sharandac/My-TTGO-Watch #370: PMU wake-up question

Crsarmv7l, 2022-08-22:
Sadly with volte being required in the US I have had to migrate away from Sailfish OS and the Amazfish companion app to android and gadgetbridge and I realized Gadgetbridge doesn't cache notifications and send on connect like Amazfish does.

With that significant limitation requiring BT to stay on with the twatch, is the pmu wakeup interval used at all since it would stay in light sleep?

sharandac, 2022-08-23:
In this mode the pmu-wakeup interval should not be used. The app then has to take care of updates itself and does not get a trigger when wifi has connected. But this is a good hint that you might want to include it.

Crsarmv7l, 2022-08-23:
Cached notifications has been an open issue for gadgetbridge for 6 years: https://codeberg.org/Freeyourgadget/Gadgetbridge/issues/578

I might take a look.

I assume you use Gadgetbridge as well?

sharandac, 2022-08-23:
Yes, I also use Gadgetbridge. In the meantime, it has also become quite stable since I switched to NimBLE.

Crsarmv7l, 2022-08-23:
NIce, I finally finished porting all my stuff to your NimBLE build. Haven't had a chance to really try it out yet though.

closed 2022-08-23 by Crsarmv7l