sharandac/My-TTGO-Watch #373: tilt/double_touch/wrist no more activ in 202207211 : OK in 220531.bin   TWatch V2

picpic020960, 2022-09-05:
Bonjour ,
all in the title
the 220531.bin from TTGO Library bin folder
the 202207211 from upload 
Thanks for Help
Regards

sharandac, 2022-09-05:
Thank you for your feedback. But I can't do anything with this error description. The version information is incorrect/incomplete. After a few tests, I could not find any problems.
Please describe the problem completely with an exact description of what you have set and how. So that we can understand the problem.

picpic020960, 2022-09-05:
'update screen' :
firmware version 
  2022050801
new version 
  2022072101

after reboot , no more double touch.

backup version from TTGO library / bin / [twatch-2020-v2-220531.bin] loaded with esptool

picpic020960, 2022-09-06:
(A) twatch-2020-v2-220531.bin (from https://github.com/Xinyuan-LilyGO/TTGO_TWatch_Library/tree/master/bin/2020-v2) = firmware version 2022050801 (see on update screen)= OK for double touch

1- receive the watch V2 with factory firmware (version ?) double touch OK
2- new version 2022072101 (see on update screen ) uploaded = KO for double touch (seems me)
3- reload with esptool (A) double touch OK
4- new version 2022072101 (see on update screen ) uploaded = OK for double touch  but KO after one night
5- reload with esptool (A) double touch OK ; wait one night and tomorrow check behaviour

all this so i wish install double touch on my V1 custom watch in micropython with break power switch 

sorry for my bad english

lewisxhe, 2022-09-07:
What do you mean by double touch?

picpic020960, 2022-09-07:
after one night the factory firmware is OK for double touch
change log beetwen 2022072101 and twatch-2020-v2-220531.bin/2022050801  ?

picpic020960, 2022-09-07:
this morning twatch-2020-v2-220531.bin/2022050801 is OK as yesterday with double touch
what is the change log between twatch-2020-v2-220531.bin/2022050801 and new version 2022072101 ?

picpic020960, 2022-09-07:
"What do you mean by double touch?"  double tap : tap two times on screen watch for wake up
(my power switch of twatch V1 is breaked and i wish replace on/off feature with double tap)

dlarue, 2022-09-17:
double tap/touch to wake display works on my 2020 V1 built from latest source with PlatformIO/Linux/Cli.  I've read that leaving the T-Watch on USB charger will damage the battery so my watch battery drains each evening and I start a new charge in the morning before work so it never continues to run through the night.

closed 2022-09-24 by sharandac