sharandac/My-TTGO-Watch #374: Extending Battery Life

moriel5, 2022-09-11:
Considering that according to this Hackaday article (https://hackaday.com/2022/04/03/tshwatch-helps-you-learn-more-about-yourself/), the ESP32's electricity thirst seems to have been conquered, with the possibility of getting over 3 days on a single charge on a 130mA battery, perhaps on the T-Watch 2020 V3's 380mA battery we can expect over a week of battery life with light usage (all sensors off, no automatic wakeup) after implementing some of @pikot's code, or even better, if @pikot could give some insight into this?

pikot, 2022-09-12:
Hi! ULP coprocessor demands  RTC pins for i2c bus, according to 
[scheme](https://0.rc.xiniu.com/g4/M00/01/F8/CgAG0mD1Jw-AaclJAAfM7x6yhFw400.jpg) t-watch has i2c pins on 21-22 and I guess RTC timer  and accelerometer connected to this pins and, and unfortunately this pins is not rtc pins, sooo we can not use ulp magic for t-watch. 
Actually I have my TSH-Watch with 300mAh battery, but I struggle with Eink which refreshes 3 sec each minute on main esp32 core and it works for 3-4 days on full charge (3 seconds each minute with consumption 40mAh is too big consumption), but! I will create quick fix (refresh only once in 10 minutes at night), And I will share battery voltage graph soon on  hackaday )

moriel5, 2022-09-12:
@pikot Thanks, it is unfortunate that the T-Watch does not make use of RTC pins on the I2C bus (hopefully Lilygo will make a new revision that does do this in the future), but thank you very much for the information, as well as the sneak peek for your next update.

I have a question, is there a reason you did not go with a removable battery? Since it looks like the T-Watch's battery could be of very good use for your TSH-Watch.

Also, I would love to see an a comparison between LCD and E-Ink on your TSH-Watch in the future, after everything is completed, just out of interest.
Hopefully that would not be a burden to you?

pikot, 2022-09-14:
Hi, @moriel5 ) A soldered battery due to simplicity.  This is second solution, at first I used Jst xh 1,25mm connector bu it takes bunch of space. In my next watch version I use 400mAh, and it perfectly fit into new case, [last 3 photos on hackaday](https://hackaday.io/project/184604-tshwatch-watch-based-on-esp32/log/204435-project-photos) 

I flashed my watches with new firmware, which refreshes Eink once in 10 minutes, and shows real time if I push a button. Curious about results ) I will share battery voltage graph. and we can compare it with LCD.

moriel5, 2022-09-15:
@pikot Thanks, that does make sense, although I would go with a slightly larger watch any day, if it meant that the battery was easily removable.

I certainly am excited about your results as well, and it would be awesome to see how the E-Ink display will stack against the LCD, battery life-wise.

dlarue, 2022-09-17:
@moriel5  and @pikot I'm only getting maybe 12 hours out of the 380mA T-Watch 2020 battery when used as a daily-wear. 
I'm looking to discuss options to 1) limit recharging to once per day and 2) hack a charging post onto the back of the watch so a 3D printed base with a pogo pin and spring(contact metal frame) can be used for charging instead of the USB connector.

Is there a forum for discussing and sharing ideas such as these?  I currently have a 2 year old 2020 V1 and a new 2020 V1 T-Watch.

pikot, 2022-09-21:
@moriel5 my results for tsh-watch with 300mAh -- 7 days 9 hours (MCU reads 5 sensors (current, temperature, humidity, rtc, pedometer) every minute and refreshes  EINK screen every 10 minutes) here is the  [VCC graph](https://cdn.hackaday.io/images/5495651663790752543.png)

moriel5, 2022-09-21:
@dlarue I believe that the main areas are here and HackADay, perhaps there are also IRC and Matrix channels?

moriel5, 2022-09-21:
@pikot Thanks, that is certainly very substantial, I imagine that without the sensors running all the time, you could easily reach 10 days like that.

Also, does your E-Ink panel support partial refresh, similar to Amazon's Kindle lineup? Since I'm theory, they could eke out a bit more battery life, by only refreshing the parts of the screen containing the elements needed to be updated (such as the time).

dlarue, 2022-10-11:
Has anyone worked on profiling power consumption of My-TTGO-Watch on the LilyGo T-Watch?

Or ideas on how I would do this?

moriel5, 2022-10-11:
@dlarue That would be an idea. When I have the time, I'll try checking (and it needed, also asking) how it was done on the PineTime (I don't have a T-Watch, however I was considering it as a replacement for my PineTime), and see how to adapt it to the T-Watch.

dlarue, 2022-10-11:
That's be great @moriel5 , the T-Watch is a nice watch but with less than 12hrs of runtime it's really limiting usability.

moriel5, 2022-10-11:
@dlarue The funny thing is, that the battery life is precisely why I decided not to get a T-Watch in the end, at least as a replacement for the PineTime, since I actually want to mostly to use it as a watch, perhaps also as a MIDI player (to aid in playing sheet, I play the piano). I am one of those who asked for a Bluetooth toggle to turn Bluetooth off on the PineTime, as well as the ability to manually set the time and date there.

dlarue, 2022-10-11:
@moriel5 LilyGo really blew it not tieing the ULP up to the correct I2C device but surely there's got to be a way to sleep or turn off devices enough to get a couple of days of use as a clock/watch.

I've turned off Wifi, Bluetooth, vibe and limited screen ON time but that's not enough. Without a profiler it seems the next step is to maybe just remove functionality down to just the RTC and display and see what results along with seeing any devices have low power commands.

pikot, 2022-10-12:
> Has anyone worked on profiling power consumption of My-TTGO-Watch on the LilyGo T-Watch?
> 
> Or ideas on how I would do this?

Main Idea for power consumption profiling is try to find slow function, and make it somehow fast)  due to esp32 consumes ~40mAh ( in case cpu freq = 80mHz) every second not in sleep mode consumes bunch of battery power.  I logged every function and looked how fast it works. 
Also I added extra ina219 (with separate MCU) module between watch and battery, which shows current between watch and battery every 1/100 seconds.

closed 2022-12-01 by sharandac

miguepollo, 2023-10-11:
hi @moriel5 I see your conversatión and I think ¿Why is not using ulp and why in other issues the rtc is not working? 
if you see something advice me

pikot, 2023-10-11:
@miguepollo , sure! you can create your own driver for bma432 and rtc and motor and touch i guess ) 
my examples you can look at - https://github.com/pikot/tsh-watch/tree/master ( main.s is enter point to all ulp logic), main logic init all drivers in c code, read all sensor data by ulp.
Ask me If you have any question about ulp

moriel5, 2023-10-11:
@miguepollo Unfortunately, I am not privy to Lilygo's decisions, so I have no idea.

moriel5, 2023-10-11:
@miguepollo That is awesome.
Now just to make sure it has a removable battery, and it may be my next watch (my PineTime's screen has cracked a bit more, and while it is still useable, I could not in good conscience continue using it as a daily driver).