sharandac/My-TTGO-Watch #375: initial build on Linux with platformIO comments

dlarue, 2022-09-12:
on Kubuntu 22.04: After installing vscode(ugh) and then platformIO from within vscode it complained platformIO failed due to missing python3-venv.  So I uninstalled platformIO, installed python3-venv(sudo apt install python3-venv) then reinstalled platformIO and that got the dev env setup.

Used git to add your project, selected watch2020-v1 ( that's what I have ) and built it.  Added my user to the dialout group( sudo adduser $USER dialout ) and uploaded the project. Worked great but there were apps for hardware not available on the 2020V1 device(GPS) so I tried editing main.cpp, commenting out #include statements( app/gps_status.... for example ) and commenting out the app setup calls lower in the main.cpp (gps_status_setup() ), saved and then hit the build button. But it didn't build anything new. I had to add a space into config.h to force a recompile to get the customized list of apps and the GPS setup application is still on the setup screens.

Thanks for doing this and I'm looking forward to getting further into it.

sharandac, 2022-09-12:
Thank you for your feedback. The GPS setup application always remains, as it also sets fake GPS via IP.
 So it is still needed.

closed 2022-09-16 by sharandac

dlarue, 2022-09-16:
One more thing if you have the time. My new unit arrived and I found it too can't run 24 hours even when it sits with display turned off.  We def need to figure out a way to improve sleep mode to improve battery drain.  Speaking of battery, I'd also would have liked to figure out a way to get a pogo pin base for charging instead of using the mini USB connector and especially since it will require multiple charges per day.  I can do software but I'm a bit better with hardware stuff like this. 
Cheers