sharandac/My-TTGO-Watch #379: Orientation screen upside down

eddovandenboom, 2022-09-24:
**Build tool used:**

- [x] Platformio (Visual Studio Code plugin)
- [ ] Platformio CLI
- [ ] Other

**used Hardware:**

- [x] TWatch 2020 or TWatch 2021
- [ ] M5Paper
- [ ] M5Core2
- [ ] WT32-SC01

**Description of problem:**
Hello! I just purchased a t-watch 2020 v3 and succesfully uploaded the code from this repository. However, the orientation of the screen is upside down.
[image]

**Additional information and things you've tried:**
After some playing around, I just solved it by adding the line `display_config.rotation = 180;` in `display.cpp` after the line `display_config.load();`. Maybe it is a mistake from the vendor, but I just thought it might help other people who encounter the same problem to post this here!

sharandac, 2022-09-24:
Thank you for your feedback! But why didn't you use the settings menu?

https://github.com/sharandac/My-TTGO-Watch/blob/master/USAGE.md

eddovandenboom, 2022-09-24:
@sharandac Thank you for your reply! Actually I just discovered the possibility to change the rotation in the display settings after I posted. I am very new to all of this (including posting on github) and it is still a lot of playing around and discovering. I don't know if this was the right place to post my observation, but it seemed that something was going wrong somewhere (even though I couldn't figure out where exactly).

For now I will close the issue since I suppose it was not a bug in your code, but a bug somewhere else.

closed 2022-09-24 by eddovandenboom