sharandac/My-TTGO-Watch #381: Interesting Pull request for Gadgetbridge

Crsarmv7l, 2022-10-28:
Came across this pull request on gadgetbridge https://codeberg.org/Freeyourgadget/Gadgetbridge/pulls/2924

Which would allow them to start using: https://developer.android.com/reference/android/companion/CompanionDeviceManager#startObservingDevicePresence(java.lang.String)

Which in turn *may* allow for the use of periodic wakes on the TTGO T-Watch, which would improve battery life.

Crsarmv7l, 2023-02-10:
Following up on this. Looks like the pull was merged here

https://codeberg.org/Freeyourgadget/Gadgetbridge/commit/0dd0b2beadd7a7172437aa77a122f521ce7c0df4#diff-a0b960e906cfbdd5845e7395aed15198e4c62768

In this file:
https://codeberg.org/Freeyourgadget/Gadgetbridge/src/commit/0dd0b2beadd7a7172437aa77a122f521ce7c0df4/app/src/main/java/nodomain/freeyourgadget/gadgetbridge/service/btle/BtLEQueue.java

I wonder if we could abuse the mPauseTransaction by setting it to true on disconnect...

Crsarmv7l, 2023-04-01:
@sharandac it's happening! Gadgetbridge cached notification pull here!

https://codeberg.org/Freeyourgadget/Gadgetbridge/pulls/3094

sharandac, 2023-04-02:
Yeah!

closed 2023-04-02 by Crsarmv7l