sharandac/My-TTGO-Watch #386: gpio: GPIO wakeup only supports level mode, but edge mode set. gpio_num:39

dlarue, 2023-01-30:
**Build tool used:**

- [ ] Platformio (Visual Studio Code plugin)
- [X ] Platformio CLI
- [ ] Other

**used Hardware:**

- [X ] TWatch 2020 or TWatch 2021
- [ ] M5Paper
- [ ] M5Core2
- [ ] WT32-SC01

**Description of problem:**
on screen sleep this error is sent out the serial port:
E (94242) gpio: GPIO wakeup only supports level mode, but edge mode set. gpio_num:39

**Additional information and things you've tried:**
in file: src/hardware/touch.cpp
line 146 -         attachInterrupt( TOUCH_INT, &touch_irq, FALLING );
I don't know if it should be set to some threshold value or what but it would seem the issue lies here.

GDV0, 2023-04-04:
Hi dlarue,
There is a fix I found in the "My-TTGO-Watch-experimental" github branch.
[image]

Just replace GPIO_INTR_POSEDGE with GPIO_INTR_HIGHLEVEL on line 418 of motion.cpp file ( located in function bma_standby). 
This prevents to have "gpio: GPIO wakeup only supports level mode, but edge mode set. gpio_num:39"

dlarue, 2023-04-04:
Have you tried it because I get an error that GPIO_INTR_HIGHLEVEL is not defined.

src/hardware/motion.cpp:418:59: error: 'GPIO_INTR_HIGHLEVEL' was not declared in this scope
             gpio_wakeup_enable ( (gpio_num_t)BMA423_INT1, GPIO_INTR_HIGHLEVEL ); //DJL

GDV0, 2023-04-04:
Oops ! Sorry, I've made a typo.
the right name is GPIO_INTR_HIGH_LEVEL

closed 2023-06-12 by sharandac