sharandac/My-TTGO-Watch #388: Compile error, missing file CreateProcess

XanCraft21, 2023-02-18:
**Build tool used:**

- [ X] Platformio (Visual Studio Code plugin)
- [ ] Platformio CLI
- [ ] Other

**used Hardware:**

- [ X] TWatch 2020 or TWatch 2021
- [ ] M5Paper
- [ ] M5Core2
- [ ] WT32-SC01

**Description of problem:**

I just got the t-watch today, and the firmware was bugged and would not boot, but when it did boot it crashed in a few seconds. I went to reflash the factory firmware using the platformio method, but it doesn’t want to upload. The error says “CreateProcess: No such file or directory”. I am using the most recent version of the software. I could not find any open issues about this topic for the most recent version, so i hope i’m not too late.

**Additional information and things you've tried:**

I tried the fix in the readme file on the front page of this repo, i tried changing the espressif version, still has the same bug.

If there is a fix or needs fixing, or which version of this repo i need to downgrade to, please let me know, and thank you for your help.

GDV0, 2023-03-24:
Hi XanCraft21
I was facing exactly the same issue. 
It seems to be a constraint due to PlateformIO on windows: file paths should not be too long !
I solved the issue with following actions:
- copy the working directoy to C:\
- modify environment name in plateformio.ini:  change **[env:t-watch2020-v1]** to **[env:twv1]** 
- modify default environment in plateformio.ini: change **default_envs = t-watch2020-v1** to **default_envs = twv1**

I finally succeeded to compile the whole application with all apps
Hope it will help !

C47hub, 2023-03-27:
Thx, this solved the problem for me to!

XanCraft21, 2023-03-28:
Thank you, i will try that soon. Can you tell me how i can do the first step?

C47hub, 2023-03-28:
Clone the git to c:\ttgo or something short like that and you should be good for the first step.

XanCraft21, 2023-03-28:
Ok. I usually don’t work with cloned git codes. Will it still work if i just change the name of the existing file?

GDV0, 2023-04-04:
Hello XanCraft21
IF you don"t want to work with cloned codes,  just download the zip file and extract the main folder ("My_TTGO-Watch-master")  from it.
After few tries, it seems useless to shorten this name.

XanCraft21, 2023-04-05:
Ok, thank you, i will try this when i can.

GDV0, 2023-04-07:
Similar to #380

XanCraft21, 2023-05-15:
Just got around to testing this and it worked, but there is a new problem now, a missing standard function round for the file My-TTGO-Watch>pio>libdeps>(Name of folder for ttgo t-watch 2020 v2)>IRremoteESP8266>src>IRac.cpp.

The standard function has no member named “round”. Is there a fix for this or can i make my own function to fix this? Thank you to whoever can give me an answer.

XanCraft21, 2023-05-15:
I just got around to trying this and it may or may not be working. It seemed that the problem i was having is not gone but i have a new problem. A missing function in the standard library for round.

In My-TTGO-Watch>pio>libdeps>(Name of the current watch you’re using)>IRremoteESP8266>scr>IRac.cpp.

The error is “round” is not a member of “std”. Is there a fix for that or could i just make my own round function? Thank you to anyone who can help me. If i manage to fix it i will add another comment if no-one has gotten back with me.

XanCraft21, 2023-05-16:
I got it compiled an uploaded. I had to make my own round function.

The software is easy to brick, and it appears to be from setting up the wifi. After re-uploading it again it remains bricked until i make my own code.

Is there anything else i could use? Should i make my own OS?

GDV0, 2023-05-19:
Hi XanCraft21,

I think you are not using the right gcc compiler version (I had the same problem when I tried to use the last compiler version)
But as it is not related to the initial problem, I recommend to close this issue (if you think it is solved)  and create a new one to discuss about round function problem

XanCraft21, 2023-05-19:
Thank you. I am not sure which compiler version i am using, i just got the latest one i could find. After making my own round function it compiles and uploads fine now.

I will close this and make another one for the round. I will need to make a second one regarding how easy it is to brick the code and how bad the boot loops are. Thank you for helping with the compile problems.

closed 2023-05-19 by XanCraft21

brucekissinger, 2023-09-08:
This solved the problem for me as well.  Thank you so much for this suggestion.