sharandac/My-TTGO-Watch #389: Go standby blocked

hermannw, 2023-03-19:
First of all, thank you for this great piece of software!
I have a similar problem with the power consumption as mentioned on other topics here. With everything off, screen dimmed almost to minimum and time out to 15 seconds, the 400mAh battery only lasts about 5-6 hours. 
Looking at the serial terminal I see this when the watch goes to sleep:

E (3772392) gpio: GPIO wakeup only supports level mode, but edge mode set. gpio_num:39
[I][powermgm.cpp:220] powermgm_loop(): T-Watch2020V3 uptime: 3733
[W][powermgm.cpp:273] powermgm_loop(): go standby blocked

...and then it stays there until woken up again.
Is it possible that something is still keeping the watch from going to sleep? If yes, what could that be?

Any help would be appreciated.
Hermann

hermannw, 2023-03-20:
Must have done something wrong... It seems to work now.

closed 2023-03-20 by hermannw

GDV0, 2023-04-07:
Hello,
I had exactly the same issue with the same TTGO T-Watch 2020 v3 400mAh and I finally found what was wrong.
standby was always false returned by this line in powermgm.cpp