sharandac/My-TTGO-Watch #391: compile error:platformio

quzx, 2023-05-23:
I git cloned this project. With vsc and platformio, i modified platformio.ini like this:
default_envs = t-watch2020-v3
During compile, I got errors like this:
.pio/libdeps/t-watch2020-v3/IRremoteESP8266/src/IRac.cpp:1131:44: error: 'round' is not a member of 'std'
     ac->setSensorTemp(static_cast<uint8_t>(std::round(sensorTemp)));
Then I added condition like this in platformio.ini:
    
[env:t-watch2020-v3]
......
build_flags = 
	-DCORE_DEBUG_LEVEL=3
    -D LILYGO_WATCH_2020_V3
    -D SERIAL_RX_BUFFER_SIZE=256
	-mfix-esp32-psram-cache-issue
    -ffunction-sections
    -fdata-sections
    -Wl,--gc-sections
    -Os
    -std=c++11
build_unflags = -std=c++98
Then execute pio init --ide==vscode
Build again.
The error is still there.
Can anyone help me? Thanks

XanCraft21, 2023-05-23:
I had this same problem. Until a fix happens i just made my own round function based off a YouTube tutorial. Do you want me to send a link to the video i found?

Also just in case i would shorten the name of the environment from t-watch-2020-v3 to tw20v3 and do the same for the default environment to prevent the missing file create process error.

XanCraft21, 2023-05-24:
I believe this question is more for quzx, but since i had the same problem i would like to put my side if that’s ok.

I just used whatever came with the packages i downloaded, that includes the watch package and the C++ and pio packages for vscode. I am not sure which one of those is missing the function.

Which version are you using? I am using the latest version.

I got a popup that asked if i wanted to add a recommended extension, have you done that on your end?

i will send my log once i get some time to do it. Thank you.

taidixiong233, 2023-05-28:
I had this same problem too.
[image]
[image]

GDV0, 2023-05-28:
Hi taidixiong233,
Could you share the beginning of the log where all software versions are mentionned ?

525bigword, 2023-05-29:
I also encountered the same problem, and I also used the environment and bracelet from t-watch version 2020-v3
Vscode version: 1.78.2
PlatformIO IDE version: 3.1.1
Here is my terminal  #log:
[image]

GDV0, 2023-05-29:
Hello 525bigword

It seems I don't have the same IRRemoteESP library installed on my side.
Yours is v2.8.5, while mine is v2.8.4.
I checked the source files of this library and I confirm there is no call to round() function in v2.8.4 but a lot in v2.8.5

525bigword, 2023-05-29:
> 你好525大字
> 
> 似乎我没有在我这边安装相同的IRRemoteESP库。你的是 v2.8.5，而我的是 v2.8.4。我检查了这个库的源文件，我确认在 v2.8.4 中没有调用 round（） 函数，但在 v2.8.5 中调用了很多

Thank you, I am a novice. How can I modify the version of the library? I found IRremoteESP8266 in the platformio.ini file, but its version is 2.7.10, as shown in the figure
[image]

525bigword, 2023-05-29:
Hello GDV0
After upgrading the IRRemoteESP version to 2.8.4, the same issue still exists
[image]

GDV0, 2023-05-29:
525bigword

In the plateformio.ini file the line IRremoteESP8266@> 2.7.10 forces the use of a version of this library > v2.7.10 and automatically loads the latest version if none is already installed.
The latest version of this library is 2.8.5 since May 8, 2023. That's probably why you have such issue.
You can test to replace the line in plateformio.ini file by IRremoteESP8266@ 2.8.4 to force the use of the correct version.

Take care: same problem could occur with other libraries

525bigword, 2023-05-29:
> 525bigword
>
> In the plateformio.ini file the line IRremoteESP8266@> 2.7.10 forces the use of a version of this library > v2.7.10 and automatically loads the latest version if none is already [...]

Okay, this issue has been resolved. It's due to the version, and it's mandatory to specify 2.8.4 IRremoteESP8266@2.8.4

XanCraft21, 2023-05-29:
Even if this did fix the bug the entire code is not usable, it bricks very easily like after a few seconds to a minute of use. I had to use one of the recommended forks, the last one being the best one so far. Thank you for your help.

525bigword, 2023-05-29:
> Even if this did fix the bug the entire code is not usable, it bricks very easily like after a few seconds to a minute of use. I had to use one of the recommended forks, the last one being the best one so far. Thank you for your help.

What is the specific issue? For example, automatic restart? Or crash?

XanCraft21, 2023-05-29:
Repeated crashes during boot. Sometimes it comes up then crashes immediately after a few seconds right back to the boot loop. I made an issue about it.

matthewbullweb, 2023-05-30:
I had the exact round issue on 2nd try. Did get it to compile once but t watch screen never came on. Just single vibration.

closed 2023-06-02 by sharandac

devnoname120, 2023-06-23:
Did you guys try my fix?
https://github.com/crankyoldgit/IRremoteESP8266/pull/1989

matthewbullweb, 2023-06-26:
Windows gave me a hard time. A raspberry pi was able to compile firmware without issue in VSCode Platform IO.

GDV0, 2023-06-27:
I confirm this fix allows to use the  actual IRremoteESP8266 library (v2.8.5)

devnoname120, 2023-07-01:
@GDV0 Could you open a pull request? As you are the author of #395.

devnoname120, 2023-07-29:
@GDV0 v2.8.6 is released and contains the https://github.com/crankyoldgit/IRremoteESP8266/pull/1989 fix. See https://github.com/crankyoldgit/IRremoteESP8266/releases/tag/v2.8.6