sharandac/My-TTGO-Watch #392: Software bricks after one use and never boots

XanCraft21, 2023-05-29:
**Build tool used:**

- [ x] Platformio (Visual Studio Code plugin)
- [ ] Platformio CLI
- [ ] Other

**used Hardware:**

- [ x] TWatch 2020 or TWatch 2021
- [ ] M5Paper
- [ ] M5Core2
- [ ] WT32-SC01

**Description of problem:**

Hello, i am having a lot of trouble getting this software to work correctly. The software becomes bricked after one use.

I have the t-watch V2 and it arrived with bricked firmware that looked similar to this one with the user interface and boot logo. It would repeatedly boot or boot loop and never work. Sometimes it would come up normally for a very short time before crashing.

I was working on making my own code, but i managed to get this one to compile, and it worked for about 2 minutes before becoming bricked again. What i had to do was use one of the recommended forks on the first page, the sixth one worked the best.

It took me ages to get the software to compile because there are more bugs than anyone can post. It looks like it has a lot of features but it’s just so buggy and glitchy.

**Additional information and things you've tried:**

I’ve tried so many things i can’t list them all here. I tried fully erasing the flash (using a different software), i tried multiple uploads, many power off and on cycles, it was still bricked.

if anyone would be willing to fix the boot loop bugs i would gladly appreciate it. Thank you for your help.

GDV0, 2023-05-29:
Hi XanCraft21

Could you share the beginning of the log when you start building the code ( the lines where all package, tool and library versions are written) ?
The sources file of the software you try to build are 6months/1year old and a lot of these packages, tools and libraries were updated in the meantime.
Unfortunately, I don't have this t-watch v2 so I can't really deeply investigate on it. 
By the way, you wrote you received your watch with a bricked firmware. Are you sure it's not "simply" a hardware issue ?

XanCraft21, 2023-05-29:
I will get the compile logs when i can, but it may take me a while to get around to it.

Can you tell me how to update the library versions? Do i just change the version numbers in platformio.ini?

I am sure the hardware in my watch is working fine. If i make my own code or use a different firmware package linked in the recommended forks it still works fine, although only the last link works.

If i also look at the serial port when i use this code a bunch of errors come from the watch. When i use the last fork link in this repo in the readme note on the first page i only get one error for a missing file but it boots fine. Only a few apps cause occasional crashes but the software doesn’t have full wifi and BLE support yet, plus no random running crashes. If you need the serial print errors too let me know.

I do also wonder if this repo is buggy because they are using an unofficial ESP32 package while i think the other one uses the official one. Although i am not sure about the package being unofficial for this one.

GDV0, 2023-05-29:
_Can you tell me how to update the library versions? Do i just change the version numbers in platformio.ini?_
**Yes** As an example if you want to avoid round function issue, replace  IRremoteESP8266@>=2.7.10 with IRremoteESP8266@ 2.8.4 in plateformio.ini

XanCraft21, 2023-06-01:
Ok, i will change that when i can. Thank you.

GDV0, 2023-06-02:
Hello XanCraft21

I'm able to reproduce your boot loop issue if I modify the plateformio.ini file to use the last TTGO_TWatch_Library (https://github.com/Xinyuan-LilyGO/TTGO_TWatch_Library.git).
Is it something you tried to do ?

XanCraft21, 2023-06-02:
No, i just used the versions that were already set. I have not gotten around to updating the libraries yet as i’ve been having a busy week. I will work on that soon.

XanCraft21, 2023-06-03:
I’m sorry, i have no time to finish testing this. Updating the libraries did not fix the boot loop problem. I’m not gonna believe my watch is defective even though it could be. This repo is badly coded and extremely buggy and unusable. I will have to find something better. Sorry to come off as rude and thank you for the help you could give me.

closed 2023-06-03 by XanCraft21