sharandac/My-TTGO-Watch #393: Multiple clicks cause the program to crash and freeze

525bigword, 2023-05-30:
**Build tool used:**

- [x] Platformio (Visual Studio Code plugin)
- [ ] Platformio CLI
- [ ] Other

**used Hardware:**

- [x] TWatch 2020 or TWatch 2021
- [ ] M5Paper
- [ ] M5Core2
- [ ] WT32-SC01

**Description of problem:**

**Additional information and things you've tried:**
Entering any app after burning the program, such as Find Phone, multiple clicks on the screen by courier can cause the program to crash
[image]

GDV0, 2023-05-31:
Hello 525bigword,

I was able to confirm the issue with FindPhone. but it seems to be linked to the task management of this app

I have done following "dirty" modifications in the code:
[image]
[image]
[image]

and I'm not able to reproduce the issue anymore.

Could you confirm on your side ?

525bigword, 2023-05-31:
Okay, after my testing, there are no issues

------------------&nbsp;原始邮件&nbsp;------------------
发件人: ***@***.***&gt;; 
发送时间: 2023年5月31日(星期三) 晚上10:56
收件人: ***@***.***&gt;; 
抄送: ***@***.***&gt;; ***@***.***&gt;; 
主题: Re: [sharandac/My-TTGO-Watch] Multiple clicks cause the program to crash and freeze (Issue #393)

 
你好525大字，
 
我能够确认FindPhone的问题。但它似乎链接到此应用程序的任务管理
 
我在代码中做了以下“脏”修改：
 
 
 
 
而且我无法再重现该问题。
 
你能确认一下吗？
 
-
 直接回复此电子邮件，在 GitHub 上查看或取消订阅。
您收到此消息是因为您创作了线程。Message ID: ***@***.***&gt;

commit 7cdb7fb2 (2023-06-02): fix #393 crashes and dep
closed 2023-06-02 by sharandac