sharandac/My-TTGO-Watch #397: Features and their effect on battery life

TimelessNL, 2023-06-13:
**used Hardware:**

- [x] TWatch 2020 or TWatch 2021
- [ ] M5Paper
- [ ] M5Core2
- [ ] WT32-SC01

**Is your feature request related to a problem? Please describe.**
Since week I've been a proud owner of a T-Watch 2020 v3 but unfortunately the battery life leaves much to be desired. I think it would be nice to have some sort of an overview of settings and their effect on battery life. As for me I only have Bluetooth (_enable on wakeup_) enabled yet my watch only lasts 7ísh hours on a charge. Of course everyone has their use case for a smartwatch, as in my case I would like to receive notifications but with it only lasting 7 hours it can't make it though a normal working day.

**Describe the solution you'd like**
An overview of settings and their effect on battery life.

**Describe alternatives you've considered**
Disabling most of the settings as wake on tilt/touch, WiFI, Location. And enabling/disabling experimental battery safe

**Additional context**
The strange thing is Gadgetbridge can still connect while Bluetooth is disabled, even after a restart of the watch. So somehow Bluetooth is not disabled and maybe other bluetooth settings may be as well and that's why the battery drains so fast? Also I do know that the battery indicator is imprecise, but is it normal that it never reaches 100%?
[image]

GDV0, 2023-06-14:
Hello TimelessNL,

Regarding the current consumption,, you are probably facing the 2 issues #386 and #389  that prohibit switching to lightsleep.
These issues have been fixed 1 or 2 weeks ago.
But I agree that having an idea about power consumption of the different options ( Wifi, Bluetooth, ...) is interesting.

TimelessNL, 2023-06-14:
Thanks for the suggestion 👍, at first I was under the impression that my watch may already have this included as I flashed master beginning of last week. But seeing that the experimental branch has been merged 5 days ago it probably wasn't 😅.

So flashed it just now and indeed the watch behaves differently, when the screen goes off Gadgetbridge looses connection so without looking at the serial log it probably means the watch goes to sleep. Unfortunately this also meant I do not receive any notifications anymore and that is really a smartwatch feature I would like have have.

I did notice @Crsarmv7l [mention](https://github.com/sharandac/My-TTGO-Watch/issues/316#issuecomment-964648203) to disable the `always on` feature and nothing would be missed, but by doing so the connection to Gadgetbridge is interrupted and no notification/calls will get through. I did also spot a `stay on` Bluetooth setting but it's unclear to me what the difference is between `stay on`  and  `always on`

Crsarmv7l, 2023-06-14:
> Thanks for the suggestion 👍, at first I was under the impression that my watch may already have this included as I flashed master beginning of last week. But seeing that the experimental branch has [...]

The reason I said that is Gadgetbridge implemented notification cache. If the watch is not connected when the notification comes in, gadgetbridge will send it the next time it connects.

TimelessNL, 2023-06-26:
I've been using the new version for 2 weeks, and must say the battery life has indeed improved.
* `Bluetooth` **enabled** and `Bluetooth always on` **enabled** = 8 hours
* `Bluetooth` **enabled** and `Bluetooth always on` **disabled** = 46 hours

This 46 hours is quite usable, unfortunately some new issues came to light.
* The clock is regularly delayed and/or minutes off. A time jump can be observed after a manual wake, sometimes even forward/backwards in multiple iterations. Ex. when waked the watch goes from 12:28(_incorrect_) -> 11:58 -> 12:43 (_correct_).
It does seem that the watch's time is synced once connected with Bluetooth after wake. But shouldn't the RTC take care of maintaining proper time during sleep in the first place?
* Maybe an issue in Gadgetbridge, but messages do not seem to be queued, sometimes they do but more often than not they don't.

ktecho, 2023-10-09:
@TimelessNL Have you found the problem with the wrong time after wake in newer versions?

My son wanted to buy the watch and put this firmware, then experiment with it, but it there is a big problem with the RTC, then maybe it's not a good choice... Thanks!