sharandac/My-TTGO-Watch #402: Hopefully T-Display-S3 TOUCH can be supported.

urobot2011, 2023-09-10:
**used Hardware:**

- [ ] TWatch 2020 or TWatch 2021
- [ ] M5Paper
- [ ] M5Core2
- [ ] WT32-SC01
- [ ] T-Display-S3 TOUCH

**Describe the solution you'd like**
I hope T-Display-S3 TOUCH is supported. It has lvgl example, screen size is 170*320.

**Additional context**
[https://github.com/Xinyuan-LilyGO/T-Display-S3/](https://github.com/Xinyuan-LilyGO/T-Display-S3/)

![](Xinyuan-LilyGO/T-Display-S3/](https://github.com/Xinyuan-LilyGO/T-Display-S3/image/T-DISPLAY-S3-TOUCH.png)

sharandac, 2023-09-11:
Thanks for your feature request. I think that should be possible without problems. But due to the lack of suitable hardware, this is not possible for me.

urobot2011, 2023-09-14:
I think I'll be able to make a pull request soon :) 
I'm currently setting up a TFT LCD driver.

closed 2023-09-14 by urobot2011