sharandac/My-TTGO-Watch #409: T-Watch S3

danielktdoranie, 2024-02-19:
I am new to ESP32 and LilyGO... I bought_ the T-Watch S3 I see this firmware doesn't support the T-Watch S3... could the T-Watch S3 be added? I updated my T-Watch S3 with the latest firmware and it looks the same... same ugly watch faces, I can see WiFi networks but not connect too them... just disappointing. If not sharandac is there any other firmware that supported the T-Watch S3? Thanks.

donniebrasc, 2024-03-15:
As do i, s3 with lora. it would be great to see a mesh implement as well but ill take usefulness

miguepollo, 2024-03-15:
I have a public fork of this repository and I am working on implement it
https://github.com/miguepollo/My-TTGO-Watch

danielktdoranie, 2024-03-15:
> I have a public fork of this repository and I am working on implement it https://github.com/miguepollo/My-TTGO-Watch

That’s awesome news! Is it ready for public testing yet

miguepollo, 2024-03-16:
No, because the clock restarts. But I'm working on getting it to work properly.

miguepollo, 2024-03-16:
If you don't see much change, I am making changes to ttgo-t-watch-libraryS3 to make it work better.

danielktdoranie, 2024-03-16:
> If you don't see much change, I am making changes to ttgo-t-watch-libraryS3 to make it work better.

Thanks so much for doing this!

crt0mega, 2024-03-16:
> If you don't see much change, I am making changes to ttgo-t-watch-libraryS3 to make it work better.

Thank you! 🥳

danielktdoranie, 2024-03-25:
> No because the clock restarts all the time. But I am working to make it work well.

Please could you let me know (maybe make a post here?) when you’re ready for beta testers? I will happily flash my T-Watch-S3 with beta firmware to help

miguepollo, 2024-03-26:
Ok I'll let you know when it's ready for the beta-testers to try it :)

ivanperez-keera, 2025-11-17:
Any updates on this?

Tagging @miguepollo .