sharandac/My-TTGO-Watch #413: t-watch2020-v3 upload error 'chip is ESP32-S3(beta3) not ESP32'

boyobejamin, 2025-02-07:
**Build tool used:**

- [ ] Platformio (Visual Studio Code plugin)
- [ ] Platformio CLI
- [ ] Other

**used Hardware:**

- [ ] TWatch 2020 or TWatch 2021
- [ ] M5Paper
- [ ] M5Core2
- [ ] WT32-SC01

**Description of problem:**

Hey folks! I'm trying to compile for `env:t-watch2020-v3`. The build works great! However the upload does not with the following log. Is there an update to any of the build tools required to compile for the detected chipset? Thank you!

```
*  Executing task: platformio run --target upload --environment t-watch2020-v3 --upload-port /dev/cu.usbmodem2101 

Processing t-watch2020-v3 (platform: espressif32@3.3.0; board: ttgo-t-watch; framework: arduino)
---------------------------------------------------------------------------------------------------------------------------------------------
Library Manager: Installing me-no-dev/ESPAsyncWebServer @ ~1.2.4
Unpacking  [####################################]  100%
Library Manager: ESP Async WebServer@1.2.4 has been installed!
Library Manager: Resolving dependencies...
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/ttgo-t-watch.html
PLATFORM: Espressif 32 (3.3.0) > TTGO T-Watch
HARDWARE: ESP32 240MHz, 320KB RAM, 16MB Flash
DEBUG: Current (esp-prog) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES: 
 - framework-arduinoespressif32 @ 3.10004.201016+sha.fb19175 
 - tool-esptoolpy @ 1.30100.210531 (3.1.0) 
 - tool-mkspiffs @ 2.230.0 (2.30) 
 - toolchain-xtensa32 @ 2.50200.97 (5.2.0)
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Library Manager: Installing me-no-dev/ESPAsyncWebServer @ ~1.2.4
Unpacking 0% 10% 20% 30% 40% 50% 60% 70% 80% 90% 100%
Library Manager: ESP Async WebServer@1.2.4 has been installed!
Library Manager: Resolving dependencies...
Library Manager: Installing Hash
Found 44 compatible libraries
Scanning dependencies...
Dependency Graph
|-- TTGO TWatch Library @ 1.4.2
|-- lvgl @ 7.11.0+sha.ec9de51
|-- AsyncTCP @ 1.1.1
|-- ArduinoJson @ 6.21.5
|-- ESP32SSDP @ 1.2.1
|-- IRremoteESP8266 @ 2.8.4
|-- PubSubClient @ 2.8.0
|-- ESP8266Audio @ 1.9.5
|-- ESP8266SAM @ 1.0.1
|-- ESP8266FtpServer @ 1.0.1
|-- ESP32-targz @ 1.0.5-beta
|-- TinyGPSPlus @ 1.1.0
|-- NimBLE-Arduino @ 1.4.3
|-- FS @ 1.0
|-- SPIFFS @ 1.0
|-- HTTPClient @ 1.2
|-- WiFi @ 1.0
|-- ESP32 Async UDP @ 1.0.0
|-- SD(esp32) @ 1.0.5
|-- HTTPUpdate @ 1.3
|-- Ticker @ 1.1
|-- twatch2021
|-- Wire @ 1.0.1
|-- Update @ 1.0
|-- ESP Async WebServer @ 1.2.4
Building in release mode
Retrieving maximum program size .pio/build/t-watch2020-v3/firmware.elf
Checking size .pio/build/t-watch2020-v3/firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [==        ]  23.7% (used 77620 bytes from 327680 bytes)
Flash: [======    ]  64.2% (used 4210565 bytes from 6553600 bytes)
Configuring upload protocol...
AVAILABLE: esp-prog, espota, esptool, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa
CURRENT: upload_protocol = esptool
Looking for upload port...
Using manually specified: /dev/cu.usbmodem2101
Uploading .pio/build/t-watch2020-v3/firmware.bin
esptool.py v3.1
Serial port /dev/cu.usbmodem2101
Connecting...

A fatal error occurred: This chip is ESP32-S3(beta3) not ESP32. Wrong --chip argument?
*** [upload] Error 2
======================================================== [FAILED] Took 12.85 seconds ========================================================

Environment     Status    Duration
--------------  --------  ------------
t-watch2020-v3  FAILED    00:00:12.846
=================================================== 1 failed, 0 succeeded in 00:00:12.846 ===================================================

 *  The terminal process "platformio 'run', '--target', 'upload', '--environment', 't-watch2020-v3', '--upload-port', '/dev/cu.usbmodem2101'" terminated with exit code: 1. 
 * 
 ```

```

**Additional information and things you've tried:**

* Attempted compilation on Windows hitting the app/ bug. Attempted on Ubuntu with the same error.

crt0mega, 2025-02-07:
Additional work is needed for the ESP32-S3 (see #409)

boyobejamin, 2025-02-08:
Oh my god. I have an S3 watch? I do! 

I've assumed this whole time it's a T-WATCH-2020-S3.

It says "T-WATCH-3" quite clearly on the box!!!! Thank you @crt0mega. This has been clearly documented in other issues. Again, thank you!

closed 2025-02-08 by boyobejamin